naubiomech/ExoCode #15: Bug in switching from bang-bang to proportional control

go226, 2019-04-13:
When walking in bang-bang and taking the baseline for proportional control, the torque setpoint breaks down and proportional control fails to function properly. Need to investigate what happens during the baseline while walking in bang-bang.

zflerner, 2019-04-13:
What happens if you're walking with a zero torque set-point in Bang Bang while taking the baseline for proportional control? 

What happens if you're in proportional control with a zero set point and you take the baseline for proportional control?

go226, 2019-04-15:
> Walking with a zero torque set-point in bang bang while taking the baseline for proportional functions as intended.

Same case for taking baseline during proportional at zero torque.

During the test with Ying today we ran into the problem of strange torque setpoints after ending and restarting the trial, plus the regular proportional control spline was strange, so we will revert the exo code and GUI to an earlier version for tomorrow.

-GO

thangnguyenac, 2019-04-15:
Correct me if I am wrong. We only take the baseline in proportional control, right? If so, then when in bang bang, we should deactivate taking baseline.

go226, 2019-04-15:
We can do that, yes. Sometimes we will take baseline while walking in bang-bang but for experiments we typically don’t do both bang-bang and proportional. We can deactivate the baseline button unless proportional is activated, certainly.

-GO

closed 2019-04-21 by go226