naubiomech/ExoCode #16: Ending trial during proportional and restarting resets the control mode to bang-bang but doesn't update GUI

go226, 2019-04-13:
When walking in proportional and ending the trial, it is expected that when the trial is restarted that the control mode is still proportional but this is not the case. The GUI indicates that proportional is active but the torque setpoint is that of bang-bang. The same control mode should remain when ending and starting trials.

zflerner, 2019-04-13:
Something in reset_leg_starting_parameters() is causing the Control_Mode to revert?

thangnguyenac, 2019-04-15:
As far as I checked, the control mode is still the same when ending and starting trials. I pulled the latest code on dev to test.

go226, 2019-04-15:
Then the bug must be somewhere else. Not a huge priority right now because of the experiments this week but I will demonstrate the error and we can diagnose from there. It may be related to other baseline issues, potentially.

We ran into this problem yesterday as well. After a torque cable was disconnected, the trial was stopped and restarted. The control was abnormal, saturating at 25 Nm even though the set point was 15 Nm. To fix this issue we deactivated and reactivated proportional control. The data from this experiment is available in Experimental Data->Overground Protocol->Unimpaired->YF->Visit 1 in the GUI folder. The file is for generation 2.

-GO

go226, 2019-04-15:
Here is an example of the control abnormality. It appears that the control
is initially in bang-bang but suddenly switches to proportional as
evidenced by the gradually increasing set point with each step (a
characteristic of bang-bang) with an abrupt shift to a response
proportional to the FSR readings.

-GO

zflerner, 2019-04-15:
I would check if the reset_leg_starting_parameters function is causing the issue by disabling it and re-testing.

Zach

Zachary F. Lerner, Ph.D.
Assistant Professor
Mechanical Engineering
Northern Arizona University
928-523-1787
biomech.nau.edu<http://biomech.nau.edu>

zflerner, 2019-04-15:
It could also be that FLAG_TWO_TOE_SENSORS resets as well in the reset leg params function.

Zach

Zachary F. Lerner, Ph.D.
Assistant Professor
Mechanical Engineering
Northern Arizona University
928-523-1787
biomech.nau.edu<http://biomech.nau.edu>

On Apr 15, 2019, at 10:55 AM, Zachary Forest Lerner <Zachary.Lerner@nau.edu<mailto:Zachary.Lerner@nau.edu>> wrote:

I would check if the reset_leg_starting_parameters function is causing the issue by disabling it and re-testing.

Zach

Zachary F. Lerner, Ph.D.
Assistant Professor
Mechanical Engineering
Northern Arizona University
928-523-1787
biomech.nau.edu<http://biomech.nau.edu/>

go226, 2019-04-23:
Removed a command in reset_starting_parameters() that returns the Two_Toe_Sensors flag to its previous condition

closed 2019-04-23 by go226