naubiomech/ExoCode #26: Improve zero torque bias issue

zflerner, 2019-05-07:
Try hard coding in the zero torque value. If that doesn't work, explore alternate approaches.

go226, 2019-05-08:
Coding the zero torque value has inconsistent effects when stopping and restarting trials. The issue is that there is an imbalance of torque on the sensor due to differences in cable tension. There is usually one position for each leg where the torque is balanced and no movement occurs unless the balance is tipped. Taking the torque calibration at different foot flexion angles has no effect on the calibration offset value. What other options should be pursued?

zflerner, 2019-05-08:
I have a feeling the torque imbalance due to the cable tension is not as strong as I felt when we were calibrating during my tests, but I’d like to confirm this tomorrow.

What’s also strange is I think we could re-take the zeroing of the torque sensors for exactly the same position (no change in tension) and get different imbalances.

My feeling is it is a voltage/ground board issue.

zflerner, 2019-05-08:
Coding the zero value and starting/stopping trials — what happens if there is no change in the angle of tension?

go226, 2019-05-08:
The left leg holds its position reasonably well while the right leg tends to gradually shift the angle with no external prodding. Adjusting the hard coded calibration value for the right leg only seems to make the foot redline faster. I have attempted several hard coded values for the right leg but no single one seems ideal based off the response. 

To confirm the grounding/voltage issue perhaps it would be best to take the calibration while the trial is running and the motors are active. 

-GO

zflerner, 2019-05-08:
It’s a chicken and egg situation. We can’t turn on the motors without finding a zero-point, yet it seems the zero-point changes with turning on the motors.

It sounds like your hard coded values were overall better than what I experienced with the zeroing average. It’s fine if it is not perfect, and a small bias exists.

Thank you,

Zach

Zachary F. Lerner, Ph.D.
Assistant Professor
Mechanical Engineering
Northern Arizona University
928-523-1787
biomech.nau.edu<http://biomech.nau.edu>

zflerner, 2019-05-08:
I would be interested in learning if the hard-code value is sensor specific.

go226, 2019-05-08:
Sure I can show you tomorrow morning before the experiment and we can decide on the course of action.

go226, 2019-05-08:
The right and left legs have different offsets but they are similar (L: 1.17, R: 1.21).

go226, 2019-05-13:
Added a feature to manually save/load the torque calibration into Teensy EEPROM memory. Now one can take a calibration while the participant is not wearing the device, save the calibration into non-volatile memory, and load the calibration during the experiment.

closed 2019-05-13 by go226